XKNX/xknx #1044: Several xknx daemons

mralessio, 2022-10-02:
For my use case I need to monitor more than one KNX/IP connections (separate lines) and periodically send the telegrams.
Is it possible to establish several tunneling connections to different KNX/IP interfaces with `deamon_mode=True` at the same time?
Thanks for help

farmio, 2022-10-02:
Not sure if `daemon_mode` works with multiple connections. Just try it out. 
Otherwise you can just call `start()` and `stop()` for every of your XKNX instance in a custom function where you wait for sigint.

mralessio, 2022-10-02:
Unfortunately, it doesn't. Next daemon will not start until the previous one has not finished. 
Not sure that just start / stop would work for me in the full scope, as I need to monitor connections constantly.

farmio, 2022-10-02:
Daemon mode doesn't do anything else than calling start() and waiting for sigint. 
Have you put your instances in Tasks? That should not block...

mralessio, 2022-10-02:
Thank you for the giving the idea. I just have started using Python recently for knx purposes. I will try to implement.

closed 2022-10-07 by farmio