XKNX/xknx #1053: KNX number platform reads value from bus without state_address configured

DirkMu, 2022-10-15:
**Description of problem:**
I have a couple of knx numbers for which the value is calculated outside the KNX bus. The value is sent from HA to the bus. For all of those KNX numbers I have not configured a state_address. However, when restarting HA there are GroupValue_Read messages on the bus. I would have expected that in that case there are only GroupValue_Write messages from HA, the restored value. 

- [X] using xknx standalone
- [ ] using Home-Assistant knx integration

**Version information:**
- xknx / Home-Assistant release with the issue: HA 2022.10.4
- last working xknx / Home-Assistant release (if known): ?

**KNX installation:**
I assume this is not relevant.

**Problem-relevant `configuration.yaml` entries (fill out even if it seems unimportant):**
```
- name: nr_gaezi_stellgroesse
  address: 4/7/15
  # state_address: 4/7/15
  type: percent
  respond_to_read: false
  min: 0
  max: 100
  step: 1.0
  mode: slider
```

**Diagnostic data of the config entry (only when Home Assistant is used)**

**Traceback (if applicable):**

farmio, 2022-10-15:
Hi 👋!

Try enabling debug log level for `xknx.state_updater` and see which exact device is scheduling the read request.
The output looks like this then:
```
DEBUG (MainThread) [xknx.state_updater] StateUpdater registered StateTrackerType.EXPIRE 60 for <RemoteValueSensor device_name="Luftfeuchtigkeit Außen" feature_name="Value" <None, 6/6/5, [], None /> />
```
Just grep the log for the group address then.

DirkMu, 2022-10-15:
Thanks for that hint. It is the climate entity that is requesting the value from the bus (command value), not the number entity. I haven't thought about them. I will close the issue.

closed 2022-10-15 by DirkMu