XKNX/xknx #1076: Failed to parse knxkeys file (latency not found)

ThyMYthOS, 2022-11-19:
**Description of problem:**

- [ ] using xknx standalone
- [x] using Home-Assistant knx integration

**Version information:**
- xknx / Home-Assistant release with the issue: 2022.11.3
- last working xknx / Home-Assistant release (if known): 2022.10.5

**KNX installation:**

* Weinzierl KNX IP Interface 732 secure
* ETS 5.7.7

**Traceback (if applicable):**

```
2022-11-19 10:38:59.531 ERROR (KNX Interface) [xknx.core] There was an error during loading the knxkeys file.
Traceback (most recent call last):
  File "/snap/home-assistant-snap/535/lib/python3.10/site-packages/xknx/secure/keyring.py", line 316, in load_key_ring
    keyring.parse_xml(dom.getElementsByTagName("Keyring")[0])
  File "/snap/home-assistant-snap/535/lib/python3.10/site-packages/xknx/secure/keyring.py", line 276, in parse_xml
    backbone.parse_xml(sub_node)
  File "/snap/home-assistant-snap/535/lib/python3.10/site-packages/xknx/secure/keyring.py", line 141, in parse_xml
    self.latency = int(self.get_attribute_value(attributes.get("Latency")))
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
```

My `.knxkeys` file only contains `<Backbone MulticastAddress="224.0.23.12" />` but no attributes for `Key` or `Latency`. Seems that those entries are optional.

marvin-w, 2022-11-19:
Fixed in https://github.com/XKNX/xknx/commit/01faeaa1128ba452ddc4f77c9c0f6e273de562d4

closed 2022-11-19 by marvin-w

farmio, 2022-11-19:
Oh dear.. so it's ETS 5 too 🤔
See https://github.com/home-assistant/core/issues/82341
I wonder how ETS has to be configured to produce the file without Latency attribute. All my tests have had it.

philipp-kuttner, 2022-11-19:
> Oh dear.. so it's ETS 5 too 🤔 See [home-assistant/core#82341](https://github.com/home-assistant/core/issues/82341)
> 
> I wonder how ETS has to be configured to produce the file without Latency attribute. All my tests have had it.
> 
> I guess `key` is only provided if Secure routing is enabled?!

Hi @farmio - I also have ETS 6 (6.0.6), not 5, if this is important to know.

marvin-w, 2022-11-19:
Maybe it is optional in some occasions (There is no schema or anything). Your fix will take care of it, however - I'd suggest that we maybe remove it from the draft PR and instead create a hotfix release to be used in HA in the meantime? @farmio what do you think?

farmio, 2022-11-19:
Yes, good idea!
We could branch off https://github.com/XKNX/xknx/commit/30ad6d3aacb6ab96f45a791f0a23dceb4e000813 to not have breaking changes.