XKNX/xknx #1108: Feature Request : DPT 235.001 intégration

bbreton09, 2022-12-18:
Hello, do you think it's possible to integrate the DPT 235.001. Is a 6bytes DPT. This contains DPT 13.010 and 5.006+1 data validity byte. 

[image]
[image]

farmio, 2022-12-18:
Hi :wave:!
Currently we don’t support “compound” DPTs as sensors. It would need a quite big change in how HA entity creation works.

bbreton09, 2022-12-18:
Hi, it's bad new :'( thank you for quick answere.

farmio, 2022-12-18:
It's not like this feature wouldn't be welcomed. It's just that someone would need to find time to implement it.

farmio, 2022-12-18:
🤔 thinking about that a little bit more, it might actually be trivial to add support for HA events and services (not sensors / entities). 
I guess that would be a good start as sensors might get created from templates easily then. 

@bbreton09 do you want to start working on that?

bbreton09, 2022-12-18:
> 🤔 thinking about that a little bit more, it might actually be trivial to add support for HA events and services (not sensors / entities). I guess that would be a good start as sensors might get created from templates easily then.
> 
> @bbreton09 do you want to start working on that?

Hi @farmio,

Unfortunately, I'm not a developer at all. I can however test things I have a machine with Hass.io and a module that supports DPT 235.001.

marvin-w, 2022-12-18:
Hi @bbreton09,

could you tell me a little bit more about your usecase with this particular DPT? What would you want to do with it?

bbreton09, 2022-12-18:
@marvin-w 

I have a Hager TX230 module which receive the active energy in kWh via teleinfo. Unfortunately this module uses DPT 235.001. Therefore active energy DPT 13.010 is combined with Tariff DPT 5.006. The goal here is therefore to be able to recover the active energy in kWh to use the HA energy module.

bbreton09, 2023-11-01:
Hi, any news about this DPT ? How can i help ?

farmio, 2023-11-03:
Hi 👋! 
No news on this front. The first answer to this issue still applies. 
Parsing the payload is not the problem - handling the result down the pipeline in xknx devices, HA entities, services and events etc. is.

closed 2024-05-26 by farmio

bbreton09, 2024-05-27:
Thank you ! Do you know how many time is take to this update is include in haos ?

bbreton09, 2024-08-09:
> It will need some changes in the integration code and the frontend too. And some testing would be good too. So definitely not in the coming release (that's in 2 days).
> 
> I did plan to release this as part of a bigger (breaking) change to xknx.

Hi, thank's your work ! I saw XKNX 3.0.0 with DPT 235.001 is release. Is he also available in HA 2024.8 ?

farmio, 2024-08-09:
Yes 👍

bbreton09, 2024-08-09:
> Yes 👍

What type should I use ? I try TariffActiveEnergy, DPT-235.001 but HA say error

https://www.home-assistant.io/integrations/knx/#sensor

farmio, 2024-08-09:
It is not supported by the sensor platform. Only by triggers (`knx.telegram` or interface device), actions (`knx.send`) and group monitor.

bbreton09, 2024-08-09:
> It is not supported by the sensor platform. Only by triggers (`knx.telegram` or interface device), actions (`knx.send`) and group monitor.
> 
> It should be possible to create a trigger based template sensor with that. https://www.home-assistant.io/integrations/template/#trigger-based-template-binary-sensors-buttons-images-numbers-selects-and-sensors

Ok, try to read in group monitor but return is null.
In ETS i assign DPT 235.001 to Address Group and upload to HA.

[image]

NHanser, 2025-05-11:
@farmio @bbreton09 : for information I opened a PR to fix this https://github.com/XKNX/xknx/pull/1685

Foulek94, 2025-10-24:
Bonjour, je suis novice dans la programmation de HA mais j’ai quelques compétences en KNX (KNX Partner). L’évolution de XkNX permet maintenant de décoder le DPT 235.001 au niveau du groupe monitor KNX. Est-ce qu’une personne pourrait m’indiquer la marche à suivre pour créer correctement le sensor afin de pouvoir l’intégrer dans l’énergie ?

Lorsque j’indique le type ou le numéro de DPT, HA refuse l’entrée indiquée.

Merci !

Hello, I am a novice in HA programming but I have some skills in KNX (KNX Partner). The evolution of XkNX now allows decoding the DPT 235.001 at the level of the KNX monitor group. Could someone tell me the procedure to follow to properly create the sensor in order to integrate it into energy?

When I indicate the type or number of DPT, HA refuses the indicated entry.
[image]

[image]

PR #1466 (2024-05-26): Add DPTTariffActiveEnergy transcoder

## Description

Closes #1108

## Type of change

- [ ] Bug fix (non-breaking change which fixes an issue)
- [x] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] Code quality improvements to existing code or addition of tests

## Checklist

- [ ] The documentation has been adjusted accordingly
- [x] Tests have been added that prove the fix is effective or that the feature works
- [x] The changes are documented in the changelog (docs/changel [... cut]