XKNX/xknx #1122: Hello, do you think it's possible to integrate the DPT 235.001. Is a 6bytes DPT. This contains DPT 13.010 and 5.006+1 data validity byte.

rtreichl, 2023-01-01:
Hello im Using HA and KNX and was just wondering why the DPT 9.009 is not pressent.
Is there a reason it is skipped?
If not can it be implemented easily?

[image]

Thanks and Bests
Richard

farmio, 2023-01-01:
Hi 👋! 

We don't have all DPTs implemented (yet). There is no particular reason. 

It can be implemented quite easily. 

On the other hand there is not really anything special about it, so you may also just use `value_type = "2byte_float"`.

farmio, 2023-01-06:
> If not can it be implemented easily?

@rtreichl do you want to do that? Here would be a good example https://github.com/XKNX/xknx/pull/397

closed 2023-01-07 by farmio