XKNX/xknx #1213: Getting the state via callback for BinarySensor

mralessio, 2023-03-04:
Hi!

I would like the `device_updated_cb` is called by force for BinarySensor and I try to use `sync()` for that. The state in fact is not changed. I thought the `ignore_internal_state=True` can be of help for this situation, but I cannot reach the desired behaviour anyway. The callback is triggered only when the value is changes to the opposite one.

```
self.upper_end_state = BinarySensor(
            self.knx, 
            'UpperEnd', 
            group_address_state = self.group_address_upper_end_state,
            device_updated_cb = self.upper_end_callback,
            ignore_internal_state=True,
            sync_state=True,
        )

# later from the method I try to use: 
await self.lower_end_state.sync(wait_for_result=True)

```
Am I doing anything wrong?
Thank you!

farmio, 2023-03-04:
Hi 👋! 
`sync()` can't be used for that, as only GroupValueWrite telegrams trigger a callback for same values (with `ignore_internal_state=True`). GourpValueResponse telegrams would only trigger it when the value changed.

mralessio, 2023-03-05:
Thank you for clarifying !

closed 2023-03-05 by mralessio