XKNX/xknx #1226: 2.6.0 filters telegrams from KNX gateway/tunnel device itself

mrdalgaard, 2023-03-15:
**Description of problem:**

Since xknx 2.6.0, xknx no longer recieves/processes telegrams originating from the KNX gateway itself.
A RTU modbus device is connected directly to the gateway, which is why the packets originates from the gateway itself (address 1.1.255)
Other telegrams from the KNX/TP bus are received fine through the gateway.

- [X] using xknx standalone
- [X] using Home-Assistant knx integration

**Version information:**
- xknx / Home-Assistant release with the issue: xknx 2.6.0
- last working xknx / Home-Assistant release (if known): xknx 2.5.0

**KNX installation:**
Using a SE SpaceLYnk gateway, with a modbus based power meter connected. The gateway exposes the modbus data via KNX address 1/1/4 as a 4byte_float.
Other devices connected to the gateway over KNX/TP are unaffected by the update to 2.6.0 and still work.

Using https://github.com/XKNX/xknx/blob/main/examples/example_telegram_monitor.py to compare the versons.
With 2.5.0 i see telegrams from the modbus device, on 2.6.0 i don't.

Adding debug logging to the same script shows the frames being recieved, but ignored on 2.6.0:
`DEBUG:xknx.cemi:Ignoring own CEMI: <CEMIFrame code="L_DATA_IND" src_addr="IndividualAddress("1.1.255")" dst_addr="GroupAddress("1/1/4")" flags="1011110011100000" tpci="TDataGroup()" payload="<GroupValueWrite value="<DPTArray value="[0xbe,0xa5,0xbb,0xb8]" />" />" />`

This change seems to be related to this commit:
https://github.com/XKNX/xknx/commit/996697ae8cce878aadfcbd95833b1fde3367008a

Is there a reason for these frames now being ignored? Is it possible to roll back this change?

farmio, 2023-03-15:
Hi 👋! 
This was initially done only for routing. Since tunnelling and routing now use a common class for CEMI processing this now affects tunnelling too. I was thinking it is ok since modern interfaces assign unique IAs for their tunnel endpoints. Apparently this is not the case for all. 
I'll remove that test - let's see how that goes.

closed 2023-03-15 by farmio

mrdalgaard, 2023-03-15:
@farmio + xknx team: Thank you for your awesome library, and fast response + fix.

I've looked at the linked issue https://github.com/home-assistant/core/issues/89581 and the LogicMachine device is basically the same as the SE SpaceLYnk device, so the same issue here. 
I've been unable to find any place to set the tunnelling IA on its web interface - Can only change its own address, but this doesn't solve anything if the tunnel IA changes to the same.

I'm also unable to get HA working with anything other then udp tunnelling on this device.

I've tested the new 2.7.0 release, and can confirm that it does indeed work again.
Let me know if you would like me to test anything else with this KNX device, and thanks again :-)

PR #1227 (2023-03-15): Don't ignore CEMIFrames from current xknx address

## Description

Don't ignore CEMIFrames from current xknx address.

Routing should be safe anyway since we have https://github.com/XKNX/xknx/blob/792864c5c8e6c9cd316af44012f8e261b5df9308/xknx/io/transport/udp_transport.py#L131

Fixes #1226
Fixes https://github.com/home-assistant/core/issues/89581

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ]  [... cut]