XKNX/xknx #1287: Home Assistant illuminance sensor shows absolutely wrong values (thousands of lux)

mikz, 2023-06-23:
**Description of problem:**

- [ ] using xknx standalone
- [x] using Home-Assistant knx integration

**Version information:**
- xknx / Home-Assistant release with the issue: HAOS  10.3, HA Core 2023.6.2
- last working xknx / Home-Assistant release (if known):

**KNX installation:**

I have GIRA 222000 Presence Detector Mini and configured it as occupancy sensor. Now I want to expose the brightness sensor in Home Assistant. I've already exposed it to HomeKit through Thinka without problems. 

Here is the ETS configuration.

[image]

Here are values from the group monitor.
[image]

Home Assistant reads it as thousands of lux, which is obviously wrong.
[image]

The current value is 297-318 lux. I configured Thinka to read the same group address and export it to homekit and it works right.
When I read the value in ETS it also responds correctly.

**Problem-relevant `configuration.yaml` entries (fill out even if it seems unimportant):**

```
sensor:
- name: "Living Room Light Sensor"
  state_address: "1/0/6"
  device_class: illuminance
  type: brightness
```

**Diagnostic data of the config entry (only when Home Assistant is used)**

**Traceback (if applicable):**

I'd be able to create a PR if I get some pointer where to look at.

farmio, 2023-06-23:
Hi 👋! 
Use `type: illuminance`.  
`type: brightness` for an DPT 9 illuminance sensor is absolutely wrong 😉
See https://www.home-assistant.io/integrations/knx/#value-types

closed 2023-06-23 by farmio

mikz, 2023-06-23:
🤦 I see. There are two. Thanks!