XKNX/xknx #1304: Error: KNX bus did not respond in time (2.0 secs) to GroupValueRead

myxor, 2023-07-30:
**Description of problem:**

- [ ] using xknx standalone
- [x] using Home-Assistant knx integration

**Version information:**
- xknx / Home-Assistant release with the issue: 2023.7.3
- last working xknx / Home-Assistant release (if known):

**KNX installation:**

No GroupValue_Read can be seen on Bus Monitor, only GroupValue Writes by the Actor itself.

**Problem-relevant `configuration.yaml` entries (fill out even if it seems unimportant):**

```
knx:
  sensor:
    - name: "Spühlmaschine Strom"
      state_address: "0/0/8"
      type: curr
      sync_state: every 20

```

**Diagnostic data of the config entry (only when Home Assistant is used)**

```
"data": {
    "xknx": {
      "version": "2.11.1",
      "current_address": "0.0.240"
    },
    "config_entry_data": {
      "individual_address": "0.0.240",
      "local_ip": "192.168.0.10",
      "multicast_group": "224.0.23.12",
      "multicast_port": 3671,
      "rate_limit": 0,
      "route_back": false,
      "state_updater": true,
      "telegram_log_size": 50,
      "connection_type": "routing_secure",
      "device_authentication": null,
      "user_id": null,
      "user_password": null,
      "tunnel_endpoint_ia": null,
      "knxkeys_filename": "knx/keyring.knxkeys",
      "knxkeys_password": "**REDACTED**",
      "backbone_key": null,
      "sync_latency_tolerance": null
    },
```

**Traceback (if applicable):**
```
2023-07-30 14:19:01.806 DEBUG (MainThread) [xknx.state_updater] StateUpdater reading 0/0/8 for Spühlmaschine Strom - Value
2023-07-30 14:19:01.806 DEBUG (MainThread) [xknx.telegram] <Telegram direction="Outgoing" source_address="0.0.240" destination_address="0/0/8" payload="<GroupValueRead />" />
2023-07-30 14:19:01.807 DEBUG (MainThread) [xknx.cemi] Outgoing CEMI: <CEMIFrame code="L_DATA_REQ" info="CEMIInfo("")" data="CEMILData(src_addr="IndividualAddress("0.0.240")" dst_addr="GroupAddress("0/0/8")" flags="1011110011100000" tpci="TDataGroup()" payload="<GroupValueRead />")" />
2023-07-30 14:19:01.808 DEBUG (KNX Interface) [xknx.knx] Encrypting frame: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="ROUTING_INDICATION" TotalLength="17" /> body="<RoutingIndication cemi="2900bce000f00008010000" />" />
2023-07-30 14:19:01.810 DEBUG (KNX Interface) [xknx.knx] Sending to 224.0.23.12:3671: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="SECURE_WRAPPER" TotalLength="55" /> body="<SecureWrapper secure_session_id="0" sequence_information="00007171d219" serial_number="0000786b6e78" message_tag="31e4" encrypted_data="b9c738864e437643c57f1531a9302051cb" message_authentication_code="d5d0866af782c2b251f0eb14d102db60" />" />
2023-07-30 14:19:01.813 DEBUG (MainThread) [xknx.cemi] Incoming CEMI confirmation: <CEMIFrame code="L_DATA_CON" info="CEMIInfo("")" data="CEMILData(src_addr="IndividualAddress("0.0.240")" dst_addr="GroupAddress("0/0/8")" flags="1011110011100000" tpci="TDataGroup()" payload="<GroupValueRead />")" />
2023-07-30 14:19:03.807 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time (2.0 secs) to GroupValueRead request for: 0/0/8
2023-07-30 14:19:03.810 WARNING (MainThread) [xknx.log] Could not sync group address '0/0/8' (Spühlmaschine Strom - Value)
```

farmio, 2023-07-30:
Hi 👋! 
Did you set up a dummy device in ETS, add the GA and reprogram your ip router / couplers?
Is the GA 0/0/8 secured or plain (DataSecure)?

myxor, 2023-07-30:
I created the group address via ETS and reprogrammed my IP Router's Email interface.
I did not set up a dummy device for this GA.
The group address security in ETS is set to "automatic" and values are written to it by the actor in a regular interval but xknx is not able to read it from the bus.

farmio, 2023-07-30:
Im not sure what you mean by "Router's email interface". 

If you set "group address security" to automatic, you should be able to see if it is in secure mode or not in the group addresses view. 

Anyway, try to set up a dummy device or consider using tunneling instead of routing.

myxor, 2023-07-30:
Indeed i switched to tunneling connection and now it seems to work.
Thanks for your help! Very appreciated.

farmio, 2023-07-30:
Have a read about "filter tables" in Knx, and how they are propagated from ETS if you want to find out why routing didn't work. Eg https://support.knx.org/hc/en-us/articles/115003363245-Filter-tables

TLDR: if the router doesn't know an incoming telegrams source may communicate with a GA, it will not route that telegram to its subnet (TP).

closed 2023-07-30 by farmio