XKNX/xknx #1352: Make devices re-attachable

farmio, 2023-11-03:
Currently the instantiation of an xknx Device object auto-attaches it to the xknx devices list.
https://github.com/XKNX/xknx/blob/a51e2b449f405cc27154b4401604b96959ee84be/xknx/devices/device.py#L45

If `Device.shutdown()` is called later, it is removed, its RemoteValues and Tasks are destroyed and it is essentially not re-usable.
https://github.com/XKNX/xknx/blob/a51e2b449f405cc27154b4401604b96959ee84be/xknx/devices/device.py#L56-L62

We should decouple the instantiation from starting and stopping tasks, so that we can stop and later re-use an instance of a Device  if we still have a reference. Like pausing a device.
- remove pending tasks
- remove callbacks (eg. `process()` for processing further telegrams)
- remove its RVs from StateUpdater
- maybe clear the state
- have a method to restart / reattach it to work again

In HA this would be needed eg. when entity_id was changed - the xknx device instance is created in `Entity.__init__()` but when an entity_id is changed, `async_remove_from_hass` and `async_added_to_hass` are called. Unfortunately there is no way to determine if the Entity object will be re-used in the `async_remove_from_hass` call. https://github.com/home-assistant/core/blob/89a9e6c6e842e7cf2265238d57ca011ed44d225e/homeassistant/components/knx/knx_entity.py#L46

Related issues
- https://github.com/home-assistant/core/issues/103116
- https://github.com/home-assistant/core/issues/82158

Bingo2023, 2024-03-08:
I think I run into this problem. KNX was runing fine until some HA update caused problems then I installed some backup.. so went back & forth with HA-Versions.

Since than KNX is unusable. I see all the knx data in Groupmonitor.
But all enties disappeared/are empty.

Any idea of how to recover from this state?

farmio, 2024-03-08:
@Bingo2023 Hi 👋! I don't think that your problem is related. 
 
This issue doesn't apply to HA restarts or integration reloads since a new xknx instance is initialized then. 
AFAIK in HA this issue is currently only applicable when changing an entities entity_id. 

Have a look at the logs - maybe they there is some hint about what's wrong in your installation.

PR #1475 (2024-06-09): Remove auto-adding device instances to `xknx.devices`

## Description

- Don't add `Device` instances to `xknx.devices` from their initializer. This starts asyncio tasks and should not happen in an `__init__` call (it could run in an non-async context).
- Devices can be attached to xknx via `xknx.devices.async_add(device)`
- Devices can be removed from xknx via `xknx.devices.async_remove(device)` 
- Devices can be re-attached. This was not possible before.

Closes #1352

## Type of change

- [ ] Bug fix (non-breaking change which fixes an issue)
- [x] New feature (non-breaking change which add [... cut]
closed 2024-06-09 by farmio