XKNX/xknx #1357: Can not Connect

henrikback, 2023-11-14:
Hello

I’m fairly new to Python programing, but I have a problem with XKNX.. I have a large house with KNX and 3 ABB IPR routers ( https://new.abb.com/products/2CDG110061R0011/ipr-s-2-1?DK ) and a 1home and a spacelogic. I have installed the XKNX lib and try to run the simpel turn on light sample in the readme. I have changed the Group address to an actual lamp in my house.

I have made a pip install xknx, and im not using home assistant.

I get this error :

L_DATA_CON Data Link Layer confirmation timed out for <CEMIFrame code="L_DATA_REQ" info="CEMIInfo("")" data="CEMILData(src_addr="IndividualAddress("1.3.241")" dst_addr="GroupAddress("1/0/31")" flags="1011110011100000" tpci="TDataGroup()" payload="<GroupValueWrite value="<DPTBinary value="False" />" />")" />

I think  i somewhere have to tell the program, to connect to one of my ABB routers? 

I have read about a xknx.yaml file, but I can not find the file ??

Hope I can get some help ?

Henrik

farmio, 2023-11-14:
Hi 👋! 
Have a look at https://github.com/XKNX/xknx/blob/main/examples/example_manual_connection.py for defining the connection.

henrikback, 2023-11-15:
thanks. Now I have a connection. Anyone know if there is a function to decode the telegram. I'm using the monitor example and editing there telegram_received_cp:

async def telegram_received_cb(telegram):
    """Do something with the received telegram."""
    print(f"Telegram received: {telegram}")

is there a function so I can get the address and the value from that telegram

farmio, 2023-11-15:
Have a look at the `tools` or `devices` packages. 
I think you'll also find examples for those.

closed 2023-11-18 by farmio