XKNX/xknx #1363: Feature Request: Add custom preflight for devices for enable tunneling

mathias-goebel, 2023-11-27:
First of all: Thank you very much for providing this piece of software! This is great! 🚀 

I want to make a feature request, since i encountered the following: I am using an IP Gateway type Weinzierl BAOS 777. This is a device also sold as OEM version under different names ("1Home KNX Bridge", "theben LuxorLiving IP1"), so i expect the feature request not exclusively targeting the original device but also its derivatives.

To get a working UDP Tunneling connection, the following steps are required before xknx is sending a connection request:
1. login at the device via REST (GET) to get an access token for further requests
2. set a specific value via REST (PUT serveritem 22 aka "TunnellingEnabled" value true)

This will authorize the client IP address to be target of a connection response.
So far it is unclear to me, when this session ends, but a restart of the device will disable all tunneling requests and the steps have to be repeated.

I like to have these two requests included as a custom script to run before any new connection attempt and in case of a loss of connection. While i am able to add this to a custom python script utilizing xknx, i can not in Home Assistant without outsourcing xknx to a custom component (and making me responsible for updates). A possible workaround can be a cronjob, but a more native support would be better for me.

References:
Documentation of the REST endpoints: https://weinzierl.de/images/download/documents/baos/knx_ip_baos_restservices.pdf

farmio, 2023-11-28:
Hi 👋! 
Are you sure this isn't an ETS setting?
According to the manual, the default configuration is to enable all 8 tunnels. I also know to have some Weinzierl 777 users, but never heard of needing to enable Tunnelling before a connection. 

Do you know if this Rest services are part of KNX specification?

mathias-goebel, 2023-11-28:
Hm, i reverse engineered this and suspect the reason is to have a security layer in front of UDP. Having a KNXnet/IP gateway exposed to all users in a home network without any further isolation is a threat.

I am using the theben IP1 and there is no product database for this device. Also the number of allowed tunnels is set to 8 here as well (read-only setting, i guess), but it is a different setting.

- [ ] check ETS product settings for 777
- [ ] check num of tunnel is read-only setting

marvin-w, 2023-12-02:
Hey! I have a Weinzierl BAOS 777 as well and I did not have to do any of those things. Maybe you can reach out to the manufacturer, but I don't think we should include something like this in xknx.

closed 2023-12-03 by mathias-goebel