XKNX/xknx #1419: DPT deoding

lukaszzyla, 2024-02-24:
**Description of problem:**

Integration seem to work fine, but several DPT are not being translated resulting in payloads in hex not being translated to values.
[image]

This doesn't seem to affect gira 222 50 motion sensor which reports fine but does affect MDT SCN IP100.03 router which sends data properly (and also includes proper DPT in payload) but integration does not translate it to vakues:
[image]

- [ ] using Home-Assistant knx integration
ETS 6 with latest apps.

farmio, 2024-02-24:
Hi 👋! 

In group monitor, as well as in device trigger, knx_event, knx.send etc. only those DPT listed in HAs sensor documentation are en/decoded. 

For date/time you can use a corresponding entity in HA to read or write.

lukaszzyla, 2024-02-24:
I have found the docs on DPT - thank you.
Anyway it is great to have KNX support in HA.

lukaszzyla, 2024-02-24:
everything works as expected.

closed 2024-02-24 by lukaszzyla