XKNX/xknx #1494: impossible to get info with DPT 13.10 DPT 235.1

jluc2808, 2024-06-19:
**Description of problem:**

- [ x] using Home-Assistant knx integration

**Version information:**
XKNX Version: 2.12.2
KNX-Frontend Version: 2024.1.20.105944
Connected to bus: Oui

**KNX installation:**
- using:  Hager domovea V2 TJA470.

my configuration.yaml
`  sensor:
# comptage général 3
    - name: "comptage général 3 puissance"
      state_address: "24/4/240"
      type: power
    - name: "comptage général 3 energie"
      state_address: "24/4/241"
      type: active_energy
      sync_state: 1`

the power is OK and return a value 
the active_energy have always status : unknown (24/4/241)

DPT of 24/4/241 is DPT13.10 DPT 235.1

farmio, 2024-06-19:
Hi 👋! 
> DPT of 24/4/241 is DPT13.10 DPT 235.1

Those are 2 different types.
If it is DPT 13.010 that is fine and you can use `active_energy`.
If it is DPT 235.001 that is just not supported (yet?) in HA. It was recently added to xknx but not yet released and it is unclear if / how these compound DPT structures can be used in sensors. 
See https://www.home-assistant.io/integrations/knx/#value-types for currently supported DPTs.

jluc2808, 2024-06-19:
i don't know what to say because in the domovea /  KNX display it is noted like i wrote 
DPT13.10 DPT 235.1
so i understand that it is confusing, but i don't have any idea of how to understand this value

jluc2808, 2024-06-19:
[image]

farmio, 2024-06-19:
🤷 sorry I don't know Domovea. Maybe have a look at the ETS project where the GA is defined... or the manual.

jluc2808, 2024-06-19:
i have the explanation for hager support (already in that post : https://github.com/openhab/openhab-addons/issues/15159)
the TE331 from hager is coded with a group address for that objet which is 6 bytes 
- 4 for DPT 13.010 corresponding to Wh
- 1 for DPT 235.01 corresponding to Tarification  
-  1 for 235.01 to control 
now the only value available in HA is a 4 bytes telegram (so the 6 bytes is unreadable) 

is there a solution to trap the 6 bytes to split in 4 bytes and let apart the 2 bytes not used ? 
or is there an evolution which will accept telegram with 6 bytes ?

farmio, 2024-06-19:
This will be possible eg. via a trigger based template sensor when the next version (to be released yet) of xknx is used in HA.

farmio, 2024-08-23:
This DPT is supported in HA 2024.8 now via `knx.telegram` trigger and `knx.send` action.

closed 2024-08-23 by farmio

jluc2808, 2024-09-04:
sorry for the question but i have always entity unknown with this device 

`2024-09-04 10:11:07.457 WARNING (MainThread) [xknx.log] Can not process <Telegram direction="Incoming" source_address="0.2.57" destination_address="24/4/243" payload="<GroupValueWrite value="<DPTArray value="[0x0,0xb8,0x74,0xca,0x0,0x0]" />" />" /> for comptage général 6 energie - Value: <CouldNotParseTelegram description="Invalid payload length for DPTActiveEnergy" expected_length="4" payload="<DPTArray value="[0x0,0xb8,0x74,0xca,0x0,0x0]" />"/>`

my definition in yaml 

`    - name: "comptage général 6 energie"
      state_address: "24/4/243"
      type: active_energy
      state_class: total_increasing`

when i monitor 24/1/243 i recieve 

`04/09/2024 10:33:18,743`

i surely miss or have a wrong declaration but i don't what or where to find 

thanks to help

farmio, 2024-09-04:
You can not use DPT 235.001 in a Knx sensor. `active_energy` refers to DPT 13.010 - which is something different. 

Use a trigger based template sensor. https://www.home-assistant.io/integrations/template/#trigger-based-template-binary-sensors-buttons-images-numbers-selects-and-sensors
with `knx.telegram` trigger.

jluc2808, 2024-09-04:
thanks for this info, but i don't know what to set if you could give me info

jluc2808, 2024-09-06:
i just look at the monitor and there is no DPT value shown for 235.001
[image]

farmio, 2024-09-06:
That's probably because the project file doesn't have the GA listed as DPT 235.001. You can check in the "Project" tab.

jluc2808, 2024-09-07:
i have checked the project tab , but it never reached the end, always with the dead circule running 
as i don't use ETS, but only Domovea (Hager) i have constructed all the mpping manually with domovea.yaml file 

here is the file .yml used 
[domovea.zip](https://github.com/user-attachments/files/16914909/domovea.zip)

and in my configuration.yaml i have added 
knx: !include domovea.yaml

farmio, 2024-09-07:
If you have no project file, you can set `type: 235.001` in the trigger configuration. 

Project data is only sourced from .knxproj, not from yaml.

jluc2808, 2024-09-07:
Unfortunately I don't succeed to find the right syntax to add the trigger, in my zip file I tried variables and add also - Plateform but I got always a wrong validation when I check to validate the yaml, so if you could give me example or syntax that will help a lot

farmio, 2024-09-08:
Something like this, but haven't tried it myself. 
```yaml
trigger:
  - platform: knx.telegram
    destination: 1/2/3
    group_value_read: false
    outgoing: false
    type: "235.001"
```

jluc2808, 2024-09-08:
doesn't work, 
if i add this sequence in knx file (yaml) i got an invalid knx syntax when i check for yaml verification 
if i add the sequence inside automation file it's OK for yaml verification but nothing append is setted 

so i don't understand how to use this trigger or this DPT in HA 

sorry for the poor value i could add

jluc2808, 2024-09-10:
thank you very much to help 
i tried what you suggest : 

added to configuration.yaml 

`
template:
  - trigger:
    - platform: knx.telegram
      destination: 24/4/243
      group_value_read: false
      outgoing: false
      type: "235.001"
    sensor:
      - name: "Trigger based template genarl 6 enegy"
        state: '{{ trigger.value.energy }}'
`

to test , restart HA 
look at monitoring Knx to find if there is velaue for 24/4/243  (i got twice in 2 minutes) 

add an entity to see what is the result with sensor.Trigger based template genarl 6 enegy

==>  result:  NONE no value is added to this sensor

jluc2808, 2024-09-10:
unfortunatly i'm like you out of idea, 
the sensor entity has been created automatically