XKNX/xknx #1550: Climate operation mode set to Sleep after HA restart

alphasixtyfive, 2024-08-19:
**Description of problem:**

- [ ] using xknx standalone
- [X] using Home-Assistant knx integration

**Version information:**
- xknx / Home-Assistant release with the issue: 2024.8.2
- last working xknx / Home-Assistant release (if known): Unsure, perhaps 2024.7.X

**Problem-relevant `configuration.yaml` entries (fill out even if it seems unimportant):**

[image]

```
  climate:
    - name: Hallway
      temperature_step: 0.5
      temperature_address: 4/6/1
      target_temperature_state_address: 4/5/1
      setpoint_shift_address: 3/3/1
      setpoint_shift_state_address: 4/3/1
      operation_mode_address: 3/1/1
      controller_status_state_address: 4/0/1
      operation_modes:
        - Comfort
        - Standby
        - Night
        - Frost Protection
```

I've tried commenting out `operation_modes` but that didn't do anything apart from adding "None" as a mode.

On start I see it is reading `[0x21]` from the bus during startup, but I'm not sure what this value translates to since this is Jung Room controller for which years ago @farmio had worked hard to add `controller_status_state_address`:

```
2024-08-19 12:23:27.783 DEBUG (MainThread) [xknx.cemi] Incoming CEMI: <CEMIFrame code="L_DATA_IND" info="CEMIInfo("")" data="CEMILData(src_addr="IndividualAddress("1.1.1")" dst_addr="GroupAddress("4/0/1")" flags="1011110011100000" tpci="TDataGroup()" payload="<GroupValueResponse value="<DPTArray value="[0x21]" />" />")" />
```

alphasixtyfive, 2024-08-19:
Thanks, it might have indeed be Julius. Perhaps it is not setting the mode? I can see the temperature change when I'm switching the mode and that temperature stays correct after reboot or config reload. I can also see following outgoing telegram when setting the mode:

```
2024-08-19 13:36:47.762 DEBUG (MainThread) [xknx.cemi] Incoming CEMI confirmation: <CEMIFrame code="L_DATA_CON" info="CEMIInfo("")" data="CEMILData(src_addr="IndividualAddress("1.1.255")" dst_addr="GroupAddress("3/1/1")" flags="1011110011100000" tpci="TDataGroup()" payload="<GroupValueWrite value="<DPTArray value="[0x1]" />" />")" />
```

alphasixtyfive, 2024-08-19:
I just rolled back to `2024.7.4` and it looks like everything is working fine, so by the looks of it the `controller_status` change actually broke it.

farmio, 2024-08-19:
Can you do a group monitor screenshot or something like that?

alphasixtyfive, 2024-08-19:
Since I'm very far away from home I'd stick with 2024.7.4 until I'm back in a few days if you don't mind. Here is the screenshot of the values when I tried cycling through presets. By the looks of it `0x21` is actually `Comfort`:

[image]

alphasixtyfive, 2024-08-19:
> Holy moly... I think I have read the bits in the wrong order when refactoring this 🙃 To fix this, a new release of xknx will be needed 😬

Thanks for figuring it out!

farmio, 2024-08-19:
Thanks for reporting 👍

PR #1551 (2024-08-19): Fix DPTHVACStatus inverted bit order

## Description

Fixes #1550

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] Code quality improvements to existing code or addition of tests

## Checklist

- [ ] The documentation has been adjusted accordingly
- [x] Tests have been added that prove the fix is effective or that the feature works
- [x] The changes are documented in the changelog (docs/changelo [... cut]
closed 2024-08-19 by farmio