XKNX/xknx #1605: Weather Station Suntracer from Elsener uses DPT 14.058 (4-byte) for pressure value

holdorr, 2024-11-24:
**Description of problem:**

- [ ] using xknx standalone
- [x] using Home-Assistant knx integration

**Version information:**
- xknx / Home-Assistant release with the issue: 3.3.0 / 2024.11.3
- last working xknx / Home-Assistant release (if known):

**KNX installation:**

Weather Station: [Suntracer KNX sl](https://www.elsner-elektronik.de/de/suntracer-knx-sl.html)

communication object: Air Pressure DPT 14.058 (4 bytes)

**Problem-relevant `configuration.yaml` entries (fill out even if it seems unimportant):**

```yaml
  weather:
    - name: "Wetter Dach"
      address_temperature: "0/3/21"
      address_brightness_south: "0/3/11"
      address_wind_speed: "0/3/40"
      address_rain_alarm: "0/3/22"
      address_frost_alarm: "0/3/24"
      address_wind_alarm: "0/3/45"
      address_air_pressure: "0/3/23"
      sync_state: init
```

**Diagnostic data of the config entry (only when Home Assistant is used)**

**Traceback (if applicable):**

```
Logger: xknx.log
Quelle: runner.py:189
Erstmals aufgetreten: 16:07:11 (24 Vorkommnisse)
Zuletzt protokolliert: 16:31:50

    Could not sync group address '2/0/33' (Unknown - State)
    Can not process <Telegram direction="Incoming" source_address="7.0.11" destination_address="0/3/23" payload="<GroupValueWrite value="<DPTArray value="[0x47,0xc7,0xc4,0x0]" />" />" data="102280.0 Pa (DPTPressure)" /> for Schnaeggi Dach - Air pressure: <CouldNotParseTelegram description="Invalid payload length for DPTPressure2Byte" expected_length="2" payload="<DPTArray value="[0x47,0xc7,0xc4,0x0]" />"/>
```

**Comment

In issue #405 the DPT type was changed from 14.058 to 9.006 to support a weather station from a different manufacturer. Is it possible, to allow both types to support this station as well?

alxwolf, 2025-07-05:
+1 for this - Netatmo also reports 4-byte values (DPT 14.058).

When I say "pressure" as type in the yaml, I get the 2-byte (9.006) - it should be DPT 14.058.

DPT 9.006 is listed as "pressure_2byte" in the documentation.

farmio, 2025-07-05:
Alright. I think these have been very different issues. Nevertheless both seem to be solved now - so I'm going to close this 😃

closed 2025-07-05 by farmio