XKNX/xknx #1615: 网关扫描有bug

bert1020, 2024-12-05:
我自己笔记本电脑, 有线网络有个 KnxIp-Router ,无线网络同样也有个 KnxIp-Router
使用样例程序测试后 examples/example_gatewayscanner.py 会只出现有线网络内的或者无线网络内的
```
1.1.0 KnxIp-Router
  192.168.123.200:3671
  Tunnelling: UDP
  Routing: Yes
1.1.0 KnxIp-Router
  192.168.123.200:3671
  Tunnelling: UDP
  Routing: Yes
```
而且会出现两个一样的
这是ETS扫描到的
[image]

farmio, 2024-12-05:
Hi 👋! 
Thanks for reporting this issue.

Could you please add 
```py
import logging

logging.basicConfig(level=logging.DEBUG)
```
to the top of the `example_gatewayscanner.py` script and share that output?

bert1020, 2024-12-06:
```
DEBUG:asyncio:Using proactor: IocpProactor
DEBUG:xknx.log:Using local ip: 192.168.1.175
DEBUG:xknx.log:Searching on WLAN / 192.168.1.175
DEBUG:xknx.knx:Sending to 224.0.23.12:3671: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="SEARCH_REQUEST_EXTENDED" TotalLength="20" /> body="<SearchRequestExtended discovery_endpoint="192.168.1.175:58444/udp" srps="[<xknx.knxip.srp.SRP object at 0x0000024CF7315820>]" />" />
DEBUG:xknx.knx:Sending to 224.0.23.12:3671: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="SEARCH_REQUEST" TotalLength="14" /> body="<SearchRequest discovery_endpoint="192.168.1.175:58444/udp" />" />
1.1.0 KnxIp-Router
  192.168.1.56:3671
  Tunnelling: UDP
  Routing: Yes
1.1.0 KnxIp-Router
  192.168.1.56:3671
  Tunnelling: UDP
  Routing: Yes
DEBUG:xknx.raw_socket:Received from ('192.168.1.56', 3671): 06100202004e0801c0a801380e57360102001100000001af02000105e000170c00314a3801e84b6e7849702d526f757465720000000000000000000000000000000000000a020201030104010501
DEBUG:xknx.knx:Received from 192.168.1.56:3671: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="SEARCH_RESPONSE" TotalLength="78" /> body="<SearchResponse control_endpoint="192.168.1.56:3671/udp" dibs="[
<DIBDeviceInformation 
	knx_medium="KNXMedium.TP1" 
	programming_mode="False" 
	individual_address="1.1.0" 
	installation_number="0" 
	project_number="0" 
	serial_number="01:af:02:00:01:05" 
	multicast_address="224.0.23.12" 
	mac_address="00:31:4a:38:01:e8" 
	name="KnxIp-Router" />,
<DIBSuppSVCFamilies families="[DIBServiceFamily.CORE version: 1, DIBServiceFamily.DEVICE_MANAGEMENT version: 1, DIBServiceFamily.TUNNELING version: 1, DIBServiceFamily.ROUTING version: 1]" />
]" />" />
DEBUG:xknx.log:Found KNX/IP device at 192.168.1.56:3671/udp: GatewayDescriptor(
    name=KnxIp-Router,
    ip_addr=192.168.1.56,
    port=3671,
    individual_address=1.1.0
    local_interface=WLAN,
    local_ip=192.168.1.175,
    core_version=1,
    supports_routing=True,
    supports_tunnelling=True,
    supports_tunnelling_tcp=False,
    supports_secure=False,
    routing_requires_secure=None
    tunnelling_requires_secure=None
    tunnelling_slots={}
)
DEBUG:xknx.raw_socket:Received from ('192.168.1.56', 3671): 06100202004e0801c0a801380e57360102001100000001af02000105e000170c00314a3801e84b6e7849702d526f757465720000000000000000000000000000000000000a020201030104010501
DEBUG:xknx.knx:Received from 192.168.1.56:3671: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="SEARCH_RESPONSE" TotalLength="78" /> body="<SearchResponse control_endpoint="192.168.1.56:3671/udp" dibs="[
<DIBDeviceInformation 
	knx_medium="KNXMedium.TP1" 
	programming_mode="False" 
	individual_address="1.1.0" 
	installation_number="0" 
	project_number="0" 
	serial_number="01:af:02:00:01:05" 
	multicast_address="224.0.23.12" 
	mac_address="00:31:4a:38:01:e8" 
	name="KnxIp-Router" />,
<DIBSuppSVCFamilies families="[DIBServiceFamily.CORE version: 1, DIBServiceFamily.DEVICE_MANAGEMENT version: 1, DIBServiceFamily.TUNNELING version: 1, DIBServiceFamily.ROUTING version: 1]" />
]" />" />
DEBUG:xknx.log:Found KNX/IP device at 192.168.1.56:3671/udp: GatewayDescriptor(
    name=KnxIp-Router,
    ip_addr=192.168.1.56,
    port=3671,
    individual_address=1.1.0
    local_interface=WLAN,
    local_ip=192.168.1.175,
    core_version=1,
    supports_routing=True,
    supports_tunnelling=True,
    supports_tunnelling_tcp=False,
    supports_secure=False,
    routing_requires_secure=None
    tunnelling_requires_secure=None
    tunnelling_slots={}
)
DEBUG:xknx.log:Closing UDP transport.
```
你是不是默认随机选一张网卡,进行广播 ?
是不是应该把所有非本地回环地址 且 状态为UP的都要进行广播一次

farmio, 2024-12-06:
You can set the `local_ip` argument manually to define a specific interface. Otherwise we use the default multicast interface.
See https://github.com/XKNX/xknx/blob/main/xknx/io/gateway_scanner.py#L244-L246
and https://github.com/XKNX/xknx/blob/main/xknx/io/util.py#L20

bert1020, 2024-12-06:
明白 多谢

closed 2024-12-06 by bert1020