XKNX/xknx #1624: Tunnel connection Issue in Cloud

TheSundayDev, 2024-12-22:
Hi all,
thanks for this library.
I'm using it and everything is ok on local but when i move my code in cloud  (eg aws lamda) I have this error even if params are ok:

**Tunnel connection could not be established**

I try also from a  remote computer outside my netwrok and it works but when I move to cloud it doesn't work.
I think it' s network issue but the network is open.
Any suggestion? 

thanks a lot

farmio, 2024-12-22:
Hi 👋! 
I hope you know what you are doing there - having access to a KNX IP interface from the internet is almost never a good idea. That said, if the connection fails, you should usually see a stack trace revealing more info.

TheSundayDev, 2024-12-23:
hi,
thanks for your reply.
I'm just testing in a dev environment.
I have just this message:
File "/opt/python/xknx/io/tunnel.py", line 209, in _connect_request
xknx.exceptions.exception.CommunicationError: ConnectRequest failed. Status code: None

thanks

closed 2025-04-11 by github-actions[bot]