XKNX/xknx #1627: No Bug: Usage of XKNX with KNX PI HAT

brenner-tobias, 2024-12-25:
I would love to use the library / HA add-on with a local PI HAT (e.g. [OpenKNX PI HAT](https://github.com/OpenKNX/OpenKNX/wiki/OpenKNX-RasPi-HAT) or [Weinzierl KNX BAOS](https://weinzierl.de/de/produkte/knx-baos-modul-838/), which should also enable to use the PI itself as the KNX IP Interface for the ETS.

Is there a possibility to do that and if so, how?

Thanks a lot in advance!

farmio, 2024-12-25:
Hi 👋! 
xknx is a KNX-IP client. There is currently no support for other KNX (non-IP) transport layers like USB or RaspberryPi HATs (don't know what protocol this uses).
You can always use a software router like knxd or Calimero to build a KNX non-IP - IP bridge. I think there are even HA Addons to be found for both of these.

brenner-tobias, 2025-01-01:
Got it, thanks. It's a shame a KNX-HAT is not supported natively, would be quite efficient to use HA as a all-in-one solution. I might look into this if I have some spare time

closed 2025-01-01 by brenner-tobias

farmio, 2025-01-02:
Great 👍 If you have any questions, feel free to reach out!