XKNX/xknx #1635: No connection to IP interface possible but ETS6 works - NAT mode?

jankap, 2025-01-07:
**Description of problem:**

- [x] using xknx standalone
- [ ] using Home-Assistant knx integration

**Version information:**
- xknx / Home-Assistant release with the issue: latest
- last working xknx / Home-Assistant release (if known):

**KNX installation:**
- MDT IP Interface with SCN-IP000.03, IP address: 192.168.178.3, PA: 1.1.14, Secure Tunneling and secure commissioning activated (not sure what that actually is), default gateway 192.168.178.1
- internet router at 192.168.178.1
- wifi router with separate subnet, IP 192.168.0.1, connected to the internet router
- Laptop with IP 192.168.0.5 where I'm running my trials with
- ETS6 works and finds the IP interface. The Group Monitor works shows communication
- To get it working I need to enable the NAT mode, see https://support.knx.org/hc/en-us/articles/4402353231762-Connection-Manager-Detailed - I'm not exactly sure what that mode does
- I can set a lamp with PA 1/0/7 on/off with ETS6
- I can't connect Bus Monitor and Group Monitor simultaneously (but why? There should be 4 individual tunnels available?)

However, xknx can't connect using this little program:

```
import asyncio
from xknx import XKNX
from xknx.io import ConnectionConfig, ConnectionType
from xknx.tools import group_value_write

connection_config = ConnectionConfig(
        connection_type=ConnectionType.TUNNELING_TCP,
        gateway_ip="192.168.178.3",
        route_back=True,
        individual_address="1.1.14",
    )

async def test_tunnel() -> None:

    xknx = XKNX(connection_config=connection_config)

    await xknx.start()
    print("Tunnel connected")
    group_value_write(xknx, "1/0/7", True)
    await asyncio.sleep(1)
    group_value_write(xknx, "1/0/7", False)
    await xknx.stop()

asyncio.run(test_tunnel())
```

The error is `xknx.exceptions.exception.CommunicationError: Tunnel connection could not be established`. I suspect this has to do with the NAT mode of ETS6 - is there something similar in XKNX or is the error likely somewhere else?

Thanks!

farmio, 2025-01-07:
> Secure Tunneling and secure commissioning activated

You'll need to pass a `.knxkeys` file or user credentials then and use `TUNNELING_TCP_SECURE`. 

See https://github.com/XKNX/xknx/blob/main/examples/example_secure.py

NAT mode is `route_back`, but omitted on TCP anyway.

jankap, 2025-01-07:
> > Secure Tunneling and secure commissioning activated
> 
> You'll need to pass a `.knxkeys` file or user credentials then and use `TUNNELING_TCP_SECURE`.
> 
Oh wow... I tried that, too, but didn't realize that the first tunnel seems to have an user Id 2
```
connection_config_secure = ConnectionConfig(
        connection_type=ConnectionType.TUNNELING_TCP_SECURE,
        gateway_ip="192.168.178.3",
        individual_address="1.1.14",
        secure_config=SecureConfig(
            device_authentication_password="<my_device_auth_pw>",
            user_id=2,
            user_password="<first_tunnel_pw>"
        )
    )
```

works! 
> See https://github.com/XKNX/xknx/blob/main/examples/example_secure.py
> 
> NAT mode is `route_back`, but omitted on TCP anyway.
ic
> 
> > I can't connect Bus Monitor and Group Monitor simultaneously
> 
> That's normal. Bus monitor doesn't allow other simultaneous connections on most devices.
Okay, good to know!

Thank you so much for pushing me to the right direction to use TUNNELING_TCP_SECURE :)

closed 2025-01-07 by jankap