XKNX/xknx #1697: gateway_scanner does not find Devices under windows

henfri, 2025-06-30:
**Description of problem:**

- [x] using xknx standalone

gateway_scanner.py (from examples) does not find any Devices in Windows (even with admin rights)

**Version information:**
- xknx: 3.8

**KNX installation:**
1.0.219 knxd
  192.168.177.3:3671
  Tunnelling: UDP
  Routing: Yes
1.0.77 KNX IP Interface 732 secure
  192.168.177.9:3671
  Tunnelling: TCP
  Routing: No

(this is the output of the gateway_scanner.py in linux)

henfri, 2025-06-30:
I am not alone:
https://knx-user-forum.de/forum/%C3%B6ffentlicher-bereich/knx-eib-forum/2043088-vorstellung-knx-lens?p=2043563#post2043563

I have deactivated the firewall with the same result.

Do you have any idea how to further debug this?

farmio, 2025-06-30:
Oh btw, have a look at your PRs... python 3.9 won't work anyway.

farmio, 2025-06-30:
You can do a capture filter for port 3671 or do a full-take and use display filter `kip`.

henfri, 2025-06-30:
Hello,

there are no packets that come from gateway_scanner.py (I am running it from an Admin-Console). 

To cross-Check, opened ETS and did get some packets (attached)

[knx1.zip](https://github.com/user-attachments/files/20979534/knx1.zip)

In the attached log, you find up to second=291 the ETS frames and after that I closed ETS and ran gateway_scanner

[image]

I am wondering whether gateway_scanner is using another Network interface.

henfri, 2025-06-30:
I suspect, this should create that log?
```
"""Example for GatewayScanner."""

import asyncio

from xknx import XKNX
from xknx.io import GatewayScanner

async def main() -> None:
    """Search for available KNX/IP devices with GatewayScanner and print out result if a device was found."""
    xknx = XKNX(log_directory=r"C:\Users\hendr\Documents\knx_parse\xknx.log")
    gatewayscanner = GatewayScanner(xknx)
```

It creates an empty log C:\Users\hendr\Documents\knx_parse\xknx.log\xknx.log

Can I /do I need to increase the log-level? If so: how?

farmio, 2025-06-30:
Hm... Ice never used that argument 😂

Have a look at the examples, some do have debug logging activated https://github.com/XKNX/xknx/blob/main/examples/example_manual_connection.py

farmio, 2025-07-01:
No idea other than Windows firewall, sorry.